BCN3D/Cura #5: Raft - T0fix wrong location in gcode

WanaGo, 2017-10-12:
Hello,

Duplication mode, Raft being used. The T0fix is being put still in Layer0, however it should be at the start of the raft instead.
With this in the default spot, the right extruder is not printing at the start. It purges correctly, but then does not print. If you pause the print immediately and then start it again after a purge, then it will start printing correctly. Moving the T0fix to where it use to be, solves the problem.
Please see image.
[image]

guillemavila, 2017-10-16:
Thanks! We found the error and will be fixed soon

WanaGo, 2017-12-06:
This has sort of come back, 1.03.
Single colour, left extruder only print.

G1 F2100 X53.432 Y134.325 E283.87325
G1 X53.502 Y134.395
M204 S389
M205 X5.62 Y5.62
G1 F1800 E277.37325
G1 F720 Z0.4
G0 F5040 X126.487 Y213.231 Z0.4
;TIME_ELAPSED:803.552750
;LAYER:1
M140 S60
M204 S1069
M205 X10.31 Y10.31
G0 F6120 X126.487 Y213.231 Z0.6
M204 S938
G1 F1200 E0
G92 E0 ; T0fix

The result though was it retracted about 28cm after doing layer 1, and then carried on.

guillemavila, 2017-12-13:
Do you still have the full Gcode? It would be important to check as it seems there has been no layer 0

WanaGo, 2017-12-13:
Yes. Attached.
The T0fix is on line 3380.
[S-J_Green-Template.zip](https://github.com/BCN3D/Cura/files/1556931/S-J_Green-Template.zip)
Zipped the gcode file as github doesnt support the file type.

guillemavila, 2017-12-14:
Thanks James, I've reviewed the gcode and this bug is already fixed in the upcoming release dacc0fb4ac3d53411847c185f0cb4de9e8a5024a

commit 63905303 (2017-10-16): Fixes #5
closed 2017-12-14 by guillemavila