BCN3D/Cura #7: Dual Color print- Crash tool head at first color change

cyclemadkiwi, 2017-10-15:
[classic_cone_Check Code Cura 1.0.0.zip](https://github.com/BCN3D/Cura/files/1385905/classic_cone_Check.Code.Cura.1.0.0.zip)
Printed a simple 2 color cone, it did the first skirt and layer 1 for first color, did tool change and skirt for second head. Tool change to first tool head again and prints to Z 10.2, then tool change to second color and moves over the job, rapids to Z0.2.....Boom Crash!!! Print does tool head change at layer 50.

Reading the code it looks to be trying to do a whole new first layer. 442 lines later it comes to layer 51 and moves the head height to Z10.2 which is correct. 

This is a BIG ERROR!!!!

cyclemadkiwi, 2017-10-16:
After looking at all the code......every time it changes the tool Head it takes the nozzle to Z0.2 another 440 lines of walls and skin before getting back to the correct Z height. Crash at every color change. I also did a multi color Draudi (Puzzle _Draudi) and as it has both colors at each layer this error did not occur!

guillemavila, 2017-10-16:
Fixed. Thanks!

commit 16dbd446 (2017-10-16): Fixes #7
commit 0e1125c3 (2017-10-16): real fixes #7
closed 2017-10-16 by eloipr