BCN3D/Cura #12: Create  or update profile from current settings

cyclemadkiwi, 2017-10-19:
Now we can see profiles again, it is now not possible to create from modified settings! 

When you adjust an inbuilt, or a self created profile the options to create new or modify are still grayed out.

eloipr, 2017-10-20:
For me it's working fine.
If you have downloaded more than one version of the installer maybe there is a conflict between the setting files. To fix it delete the next folders:
C:\Users\<user_name>\AppData\Roaming\BCN3D Cura
C:\Users\<user_name>\AppData\Local\BCN3D Cura
If it doesn't fix the error, download the [new installer](https://drive.google.com/file/d/0B9EhngnR5bl6N1Q1cXFzTkpfazQ/view). We've made some bug fixes.

closed 2017-10-26 by eloipr

cyclemadkiwi, 2017-11-02:
Hi,

Thanks for the info, it has been working great now on 1.0.2

I'm looking forward to the official release, Ive been printing lots of
things and did a 42 hour print on the weekend that went great (After I had
to rebuild the extruder!) found a small issue with how I had assembled it.
Going to try my extruders on NinjaFlex today. Our extruders with my design
tweaks have not had any grinding at all since my update/additional part.
Bondtech extruders really are not required....we will see after tonight! :-)

Thanks team :-)

Regards
Glen