BCN3D/Cura #14: Duplication with Prevent Filament Grinding enabled

WanaGo, 2017-10-23:
Hi,
If you have Prevent Filament Grinding enabled, and then enable duplication, that option goes away, however it is still active.
When printing, and the heads return to the purge bins to purge some material, the printer goes 'mental'. Banging/grinding/loud noises come from the heads, they do not return to printing like they do after a Pause. I had to shut the printer off mid print. This is not working at all.

Also when you do a Pause, the printer does not retract after the pause, and ends up making a blob at the restart point.

Did this in 1.0.0 and also doing it in 1.0.1-beta

guillemavila, 2017-10-24:
Hello James, I assume you are using a Sigma in duplication mode.
We know this feature is not working. The Sigma officially doesn't support Mirror/Duplication, so it's a feature that's not working because you _should_ never be able to enable it.

However (now the good news):

- next firmware & Cura releases will change the Prevent Filament Grinding sequence to enable this feature apart from the print mode selected.
- if you don't want to wait, you can disable the feature manually in the bcn3dsigma.def.json changing the lines
```
        "avoid_grinding_filament":
        {
            "enabled": true,
            "value": true
        },
```
to:
```
        "avoid_grinding_filament":
        {
            "enabled": "print_mode != 'duplication'",
            "value": "True if print_mode != 'duplication' else False"
        },
```

WanaGo, 2017-10-24:
Surely this is also an issue for the SigmaX?
So this code above disables the feature in Cura, what about solving it so it works instead?
Also the pause/resume, so it retracts correctly.

guillemavila, 2017-10-24:
No. First because Sigmax come with new extruders which actually don't need this feature, and second because even if you wanted to enable the feature, Sigmax firmware is already capable of it.

As I said, it will be solved in the next release of Cura & Firmware for the Sigma, just to unify Sigma/Sigmax firmwares and gcodes. Cura will use the same protocol for the feature for both Sigma and Sigmax. Right now it has to give different commands to each one for firmware compatibility.

We'll take a look to the pause issue on the Sigmax. Note you are using a machine in a way we are not giving support. We really appreciate the contribution to the comunity but it's not something we are officially offering to our clients. Thanks!

lemmans, 2017-11-03:
Yes please, reconsider your decission of artificially castrate the original sigma, not allowing duplication and mirror modes.

closed 2017-11-07 by eloipr