BCN3D/Cura #15: Dual color print issue

cyclemadkiwi, 2017-10-24:
Just started a dual color print with ABS in both extruders, the first extruder (T0) did the correct temp for the skirt and first layer, but when T1 came out it did the skirt and first layer 10°C lower than it was told to in Cura. the rest of the layers were all at the correct temp! The M109 commands in the first layers for T1 are not correct. Auto Temp was turned off.

guillemavila, 2017-10-26:
We are not having this issue... Just uploaded the[ installer 1.0.2](https://github.com/BCN3D/Cura/releases/tag/v1.0.2-beta) which solves almost all issues reported until now, you can test it and if everything is solved I'll close the issue.

_As we are upgrading between beta releases it's recommended to delete these folders:
C:\Users<user_name>\AppData\Roaming\BCN3D Cura
C:\Users<user_name>\AppData\Local\BCN3D Cura_

cyclemadkiwi, 2017-10-27:
I have been using the 1.0.2 since Eloipr sent me some fixes....all seems to be working really well....just started a 42 hour dual print on my Sigma :-)

guillemavila, 2017-11-03:
Awesome, new version with more corrections available!

(Not needed to remove these folders manually anymore)

closed 2017-11-03 by guillemavila