BCN3D/Cura #17: User data storage location

ChrisTerBeke, 2017-11-02:
A recent issue on the Ultimaker Cura GitHub repo made us aware that BCN3D's Cura is saving user data to the same folder as Ultimaker Cura. Some people are running both version and are running into issue because of this. It would be nice if you could change the storage location of user data so both versions can run alongside each other.

Details: https://github.com/Ultimaker/Cura/issues/2709#issuecomment-341358973

eloipr, 2017-11-02:
We are saving user data in a different folder named BCN3D Cura. In our computers we have multiple versions of Cura installed and everything is working fine.
I don't know what can be causing the problem, but I'll try to find the reason.

Thank you for reporting the problem!

cyclemadkiwi, 2017-11-05:
I have been running multiple versions as well and have not had any issues. Was using 2.7 Cura and BCN3D Cura with no issues. The latest Cura however, is no good without the BCN3D flavour settings as in ver 2.7

Sticking with BCN Cura 1.0.3....it's going pretty well now!

closed 2017-11-07 by eloipr