BCN3D/Cura #20: Temperatures not correct when printing with right extruder only

IdiocyCubed, 2017-11-04:
Hi,
Using the latest pre-release to ensure this hasn't already been fixed.  The issue is when you print a part with the right extruder only and use skirt for build adhesion (haven't tried others).  If you select the right extruder for build plate adhesion so the left extruder is not used at all it doesn't change temperature to the printing temperature.  So the gcode has one call to M104 at the start which is the printing temperature initial layer (260 C) and then it never sets the temperature to the printing temp which should be 250 C in this case.  If I change it to use left extruder for build plate adhesion the temperatures are set correctly.  I can provide the profile and gcode if necessary.
Cheers

S3D-r, 2017-12-03:
I noticed the same problem with R/E (didn't check with L/E). I have adhesion type as "None". The nozzle temperature stays at initial nozzle temperature during the whole print.

[image]

guillemavila, 2017-12-12:
We're working on it,  thanks for reporting

S3D-r, 2017-12-14:
You haven't flagged this as a bug, shouldn't you?

commit 39da9cac (2017-12-20): Corrected unintentional removed temperature commands

Closes #20
closed 2017-12-20 by guillemavila