BCN3D/Cura #23: Infill shows always the same in preview

S3D-r, 2017-11-06:
Print preview shows the infill always the same, no matter the % is. Here I have 100% infill, and the infill (brown color) looks exactly same with 40% or 80% etc...

[image]

eloipr, 2017-11-07:
Have you done the slice again?
When a setting is changed, a slice is needed in order to preview it in the layer view mode.

S3D-r, 2017-11-07:
Hi!

Cura performs the slicing automatically after any change, and so did it in this case (it took <10 sec.). Before taking the screen shot, I wait until the slicing had finished. I made about five tests with different infill.

However, I today opened the same project file, and the infill view had changed, and now it looks as it's supposed to (with different infill %'s)!  

It was a late evening yesterday , Cura may have been tired and it refused to do its job and it just needed to sleep over night!  ;-)    Next time in similar case I try rebooting Cura and/or reboot PC.

closed 2017-11-07 by eloipr