BCN3D/Cura #24: Dual material Print

cyclemadkiwi, 2017-11-08:
Left head ASA Right head HIPS for support, standby temp set at 150°C
Smart Purge turned off.
the first 8 layers at every layer, the right head moves to X0.0 Y210 (Guess, middle rear of table) after each completed layer of the left head, no support was required at this stage, the right extruder sat at full temp (250°C) for the entire print, not once dropping to standby temp.
It continues to do this even after going above Z height where no support was required or existed in the layers view in Cura.

cyclemadkiwi, 2017-11-08:
today it works ok! maybe a setting wrong :-(
Cura put some support in odd place, 5 identical parts placed at identical spacing but the odd squirt of support only on 2 parts. otherwise the right extruder is now going to standby temp.

svengl, 2017-11-13:
Same issue here. Right extrudes moves to the rear center and does nothing, then goes back to the stop.

guillemavila, 2017-11-15:
Is it happening with the latest version?

https://github.com/BCN3D/Cura/releases/tag/v1.0.3

cyclemadkiwi, 2017-11-15:
Yes, it is with 1.0.3

maodch, 2017-11-19:
Got the same issue. Might be the primetower position, even there isn't one. 
like going to prime ... ohhh damn, there is no prime tower set ... going to purge tray instead :-)

svengl, 2017-11-22:
Yes happens with 1.0.3

guillemavila, 2017-12-20:
It uses to happen to us traveling to the point (`layer start X`, `layer start Y`). But we are actually fixing this unexpected travel.

Do you have some parameter pointing to those X/Y coordinates? (0, 210) in your case.

guillemavila, 2017-12-20:
Issue name changed for better order

svengl, 2017-12-21:
I couldn't reproduce it yet with other dual extrusion prints. So it is not
always.

guillemavila, 2018-03-12:
Fixed https://github.com/BCN3D/Cura/commit/f2990a9554e493b2c7723d0e4cc2744c67b4b549

closed 2018-03-12 by guillemavila