BCN3D/Cura #26: Standby temperature bug + missing feature

S3D-r, 2017-11-13:
A bug:  In dual color print you may set the the idle nozzle to stay at lower temp for saving the filament from oozing/burning. I have one experience only, but it work like this:
(RE = right extruder, LE = left):  Printing work starts with LE. However, RE will also heated to **full temp**. It shouldn't !  When it's time for first color change, LE cools to idle temp and RE starts printing. After that, everything goes as supposed. 

Missing feature: In dual-color print, once either of the colors is printed last time (rest of the print will be one-color), the nozzle cools down to idle temp, but not to ambient temp. It's completely unnecessary to keep the nozzle heated now that it won't be used any more. Imagine: You have soluble support in the beginning (only), and rest is print material. The soluble material nozzle stays heated to the end of the print, that may be 20 hours (in a long print)!

guillemavila, 2017-11-15:
The bug shouldn't be a bug. Both hotends are heated up if we're printing with both, so you can print an skirt and make sure both hotends are properly primed and the idle one will start properly when entering to the model. Printing the skirt is highly recommended. Did you have it disabled?

We'll consider this feature, although the standby temperature should be non harmful for the filament but stay warm enough to quickly get it back to print temperature, I think it's interesting.
Thanks!

S3D-r, 2017-11-16:
I forgot to mention, but my concern applies to long printing works (10...30 hours). In case of Sigma, the priming will be done by extruding to the reservoir, not on the platform. It's insane to have the idle extruder wait for its turn in full temperature (say 220C) for (say) ten hours before first printing. I currently had a work where first 12 hours is printed black (LE) and then 30 mins white (RE), and then 3 hours of black.  The white PLA was no more white when the filament had been standing still at 220C inside the extruder...   I simply don't see any sense having the idle extruder fully hot that long time.

What comes to the feature suggestion, you apparently got me wrong: There's no reason to keep filament "waiting to quickly get it back to print temperature", as in my case the filament had already done its job and it won't get back to printing any more. It's unnecessary to waste energy and keep it warm, even if the filament could stand it.

guillemavila, 2017-11-16:
For the first case, you should set up a proper idle temperature, we set it at 150C so the material is not overheated. For the second case, you can set an idle temperature of 0C and by now will make the point.

We will look how to turn off the idle hotend when it won't be used anymore during the print.

S3D-r, 2017-11-16:
Case 1: No. As I wrote, the extruder won't go to idle temp until it has done its first printing cycle. It will go directly to full printing temp (even if it's first job is tens of hours away). So, setting idle temp to zero won't solve this problem.  

Case 2: This is no big deal, the main harm is wasted energy. Not work big effort, but if it's easily done - why not!

guillemavila, 2017-11-16:
We will take a look on it.

For both cases everything will work as you want if you print the skirt at the beginning and set up a proper standby temperature for both extruders. If not, please attach the gcode. I assume you're using [latest version 1.0.3](https://github.com/BCN3D/Cura/releases/tag/v1.0.3).

Thanks for testing!

S3D-r, 2017-11-16:
Hi!   OK, I will test with printing skirt.   

I posted this thread **3 days ago**. I was young at that time, but I can still remember those good old days.  Version 1.0.3. was released **2 days ago**, so my issue doesn't apply it. I'll give it (1.0.3) a try.

S3D-r, 2017-11-16:
An additional comment:  Printing a skirt is workaround the problem, not fixing it. I still stick in my original position and call it a bug...  However, suggesting workarounds is valuable part of this group, and I appreciate it. We're living in an imperfect world (that is very true with 3D printing...) and need to apply workarounds every now and then. Thank you!

guillemavila, 2017-11-16:
Thank you too for sharing, hopefully we'll solve it in the next release ;)

guillemavila, 2017-12-20:
Issue name updated for better order

guillemavila, 2018-01-16:
Hotends are turned off if not used anymore daacad320d0ab21b92860f8ffd0474b09ee9f459

guillemavila, 2018-01-17:
Standby temperature works as expected d0e309c34bc0c07e932e5f87689cde2fe254b46b

closed 2018-01-17 by guillemavila