BCN3D/Cura #28: Skirt not going around all parts

WanaGo, 2017-11-15:
I have noticed this a few times now. The skirt does not always go around all the parts, but can cut through some, like in this example.

[image]

Unsure what is causing it

Model from here: https://makeprintable.com/printables/download/586e24e2d13f4a6dda5bd7a9

Regards

eloipr, 2017-11-15:
[image]

The first layer of the sheep legs are not touching the platform. 
Enabling support or translating the model to Z -1 fixes the problem.

WanaGo, 2017-11-15:
Ah, yes you are right. Also flipping it over 180 does the trick. the other side is flush. Seems the legs are a fraction shorter than the body, and it imported upside down. Thanks

closed 2017-11-15 by WanaGo