BCN3D/Cura #31: display errors

hst68, 2017-11-16:
In the current version 1.03 for macOS 10.13 there are display errors. See attachment
[image]

hst68, 2017-11-16:
[image]

hst68, 2017-11-18:
My Solution:
Go to Applications > Font Book and doing File > Restore Standard Fonts and restarting.

You could also try to clean the font cache:
Terminal
sudo atsutil databases -remove

eloipr, 2017-11-20:
Thanks for sharing the solution!

hst68, 2017-11-20:
unfortunately the problem persists partially

eloipr, 2017-11-20:
I've also experienced this problem using cura in a virtual machine. Don't know what can be causing it.

atlemo, 2017-11-20:
This is a known issue with Cura, apparently:
https://ultimaker.com/en/community/51334-cura-fonts-garbled-on-macos-high-sierra

guillemavila, 2018-04-10:
Fixed b4b7691b44e01dd600382754e41b07892d0b1cd1

closed 2018-04-10 by guillemavila