BCN3D/Cura #36: BcnCura 1.0.2 bug

yuchu3d, 2017-11-22:
i can't select material in BCNCura 1.0.2

eloipr, 2017-11-22:
Download the latest release 1.0.3.

yuchu3d, 2017-11-22:
I installed the 1.0.3 but  it doesn't work. Must i unstall the 1.0.2 before installing 1.0.3?

------------------
Name: Yuchu Huang
Tel: 13553538492
Adress: Shiqiao street Dongxing path the number of 476 in Canton city Guangdong province (广东省广州市番禺区市桥街东兴路476
 号星海通信息文化产业园）

yuchu3d, 2017-11-22:
i unstalled 1.0.2 and reinstalled 1.0.3 but no use, i still have the same problem

eloipr, 2017-11-22:
Are you using windows, mac or linux?

yuchu3d, 2017-11-22:
Windows

------------------
Name: Yuchu Huang
Tel: 13553538492
Adress: Shiqiao street Dongxing path the number of 476 in Canton city Guangdong province (广东省广州市番禺区市桥街东兴路476
 号星海通信息文化产业园）

eloipr, 2017-11-22:
Try deleting these folders:
C:\Users\<user_name>\AppData\Roaming\BCN3D Cura\
C:\Users\<user_name>\AppData\Local\BCN3D Cura\

Maybe there are conflicts in the cura configuration files between versions.
It shouldn't be happening in the latest versions...

yuchu3d, 2017-11-27:
Dejectedly, I deleted the two files, but it still doesn't work.

------------------
Name: Yuchu Huang
Tel: 13553538492
Adress: Shiqiao street Dongxing path the number of 476 in Canton city Guangdong province (广东省广州市番禺区市桥街东兴路476
 号星海通信息文化产业园）

eloipr, 2017-11-27:
Send me this file to epardo@fundaciocim.org.
C:\Users\<user_name>\AppData\Roaming\BCN3D Cura\1.0.3\BCN3D Cura.log

closed 2017-11-27 by eloipr