BCN3D/Cura #38: Prime Towers seperated for each material

maodch, 2017-11-30:
Having the Issue, that when printing PVA with any other Material, the no material really stick to PVA or vice versa.

So, I tried to have 3 Objects: Left= PRIME 1 (Extruder 1), Middle (Target Dual Color Object), Right PRIME2 (Extruder 2). 

This consteallation don't work, becuase I'm not able to tell Cura, which order the objects sould be made.

Is there such a function I didn't see or is it really not possible.

After reading may posts in the www all having the same issue when printing with support material like PVA or HIPS.

I don't understand which criterias are taken for printing in the order cura does.

Any suggestion how to solve my issue?

Thanks in advance
Ramon

guillemavila, 2017-12-14:
No, there's no option like that in Cura. Things I would try:

- Bigger tower, so the material has more surface to stick
- Enable SmartPurge and tower,so you avoid underextrusion at the beginning of the tower. Be careful as smartpurge will be activated in both extruders (known bug)
- Insert two block shapes (stl) in front of the part and assign each one to one extruder. I haven't tested it, but Cura should print first the objects closer to the front of the bed.
- Enable ooze shield and tower, so you purge in the tower, and if it falls, the shield will stop the residual to hit the part

maodch, 2017-12-16:
@guillemavila  thanks for the tipp, but unfortunately the way of working is quite different.
I tried any Settings (before, behind, left, right, at the very bottom etc.), but no will really help.
Same for Simplify.

Thanks anyway

guillemavila, 2017-12-18:
Ok, I understand, for S3D you can test a workaround with 3 processes to print:
First layer of the prime towers 0.19mm
First layer of the part 0.2mm
Consequent layers for all parts the same size. This way towers will always have an offset of -0.01mm and will print first. I've tried it and doesn't seem to work as we'd like, but probably playing around with different processes and solids you can make it work. 
We will take it as an enhancement proposal for newer versions of Cura.

maodch, 2017-12-18:
thanks, I give it a try and come back to you :-)

maodch, 2018-01-06:
@guillemavila Your suggestion is working (more or less). Most of the time the prime tower will be prefered to print first, but 30% of the time not. Not sure why, maybe roundings or what ever.
My guess ist, that every time, the process stratetd to print 1 Outline, 1 Infill, 2 Ountline, 2 Infill go to next Object, the priorisation failse, because he has been on the same object and for the slicer it makes no sense to go to anoter object fist, because of wastetd time.

So it would be really cool if you take this into furter releases

guillemavila, 2018-01-08:
We'll keep it in mind ;)