BCN3D/Cura #40: Hotend temps in gcode not reflecting settings in BCN3D Cura 1.0.3

billynomates, 2017-12-05:
I have specified two custom filaments by duplicating definitions of existing filaments in the 'Manage Materials' tool then adjusting the required printing temperature. I then unlinked the new definition.
One is a PVA dissolvable support from Form Futura recommended print temp 190-210, the other a Taulman nylon recommended print temp 230.

When I try to use these filaments, the gcode that BCN3D Cura 1.0.3 produces does not reflect the temperature settings that I have defined.

Moreover when I search through the gcode the temperature setting migrates up and down through the slices. Now I understand that I can tell it to do that to some degree, but I'm not telling it to migrate and that migration is resetting the hotend away from the desired temperature at every new slice. For the PVA it jumps to 220C and for the nylon it jumps away from the 230 range and up to the 250-260 range where most nylons sit. 

Basically the slicer seems to be ignoring the filaments that I'm telling it to use and refering to the originals that were the starting point for the new definitions.

MacOS 10.12.6
Sigma R17

ZAMHelmets, 2017-12-05:
I have the same experience for the following settings:
- temperature, printing, initial printing and initial layer
- retraction speeds
- print speeds
- whether or not to use smart purge
- Anecdotally there are other parameters that I question, but haven't had time to track down. In general it almost seems like overriding an out of the box profile doesn't necessarily change anything.

I make the change in BCN3D Cura, generate new code and then have manually inspected the code to verify the changes.

guillemavila, 2017-12-14:
Do you have auto temperature parameter enabled? This automatically changes temperatures according to layer time, flow extruded and some other parameters. Next Cura release will come with the feature disabled by default as it's giving more confusion than anything.

Speeds shouldn't have any problem, I'm taking a look at it.

Smartpurge has a bug and applies to both extruders if it's enabled in either one or both. To disable it, it must be unchecked in both extruders. Will be fixed in the next release

billynomates, 2017-12-14:
Thanks G,
I did have autotemp checked, but having now unchecked it and re-tried the same issue still happens.
I'm setting Extruder One/Material One to 230C through the UI, slicing and the g-code says 260, etc, etc.
Does the autotemp carry over from the profile into the definition of the custom material perhaps ?
I haven't tried creating the custom material again without autotemp checked, but I will.

billynomates, 2017-12-14:
No luck.
I tried turning off autotemp and recreating the custom filament the same way.
I get the same results in the gcode.

ZAMHelmets, 2017-12-14:
I've also disabled auto temp (too many issues) and found that you have to set all the temperatures (initial, final, print, etc.) the same to get consistent results. There's also an issue with standby temperature being bypassed if the 2nd extruder does not run early in the print. The standby temp is ignored and then activates midpoint.

guillemavila, 2017-12-15:
SmartPurge activation fixed 2da68bbd9f11dfe3c2a77d253d3afb19c434ba5d

guillemavila, 2017-12-20:
@ZAMHelmets 
- You have to change all temperatures to the same value if you need to print at the same temperature always. Auto temperature disabled.
- Standby temperature will only be triggered after the hotend has printed something. So it's highly recommended to start the print with a skirt, which will also allow you to double check a proper calibration and extrusion for each hotend. After the skirt, the hotend will be able to cool down if idle.

ZAMHelmets, 2017-12-20:
Thanks guillemavila. That makes sense. I've been just purging the hotends and avoiding skirts to save time (and there isn't always space when printing larger parts). Since the printers have done a good job staying calibrated, this has worked well up until recently.

guillemavila, 2017-12-21:
[New beta](https://github.com/BCN3D/Cura/releases/tag/v1.1.0-beta) released!
Ideally with all these issues fixed ;)

closed 2018-06-13 by eloipr