BCN3D/Cura #41: Provide instruction to build on Linux

asyd, 2017-12-06:
Hi,

I'm wondering if you can provide instruction to build cura-sigma on Linux? Or at least help me? Should I use `cura-build` repository from `Ultimaker` on this repository? I tried to use `cmake` on this reposotiry but according to the `Jenkinsfile` it's only to run tests. 

If you help me I'm ok to write the doc :)

Thanks

eloipr, 2017-12-11:
Ultimaker has 2 repos to build Cura (cura-build-environment and cura-build). First you need to build cura-build-environment. Since BCN3D Cura is based on Cura 2.7 you'll have to make some changes. Change GIT_TAG origin/master to GIT_TAG origin/2.7 in projects/arcus.cmake and projects/savitar.cmake.
Then you'll have to build Cura with cura-build repo.
We have a fork of cura-build but I recommend you to use the Ultimaker project first and once you know how it works you can use ours.
This process can be really tricky, if you get stuck I'll be glad to help ;)

asyd, 2017-12-13:
Thanks for your answer!

I'm working on it.. 

Have you planned to release linux releases btw? :)

eloipr, 2017-12-14:
There is a linux release. It is an AppImage

eloipr, 2017-12-14:
You can download it [here](https://github.com/BCN3D/Cura/releases/download/v1.0.3/BCN3D.Cura-1.0.3.AppImage)

closed 2018-04-12 by eloipr