BCN3D/Cura #43: Cura swaps left and right temperature settings?

S3D-r, 2017-12-14:
I'm printing with the LEFT. 

Printing starts at 225C, but at some point (didn't follow, 12 hours print) goes down to 207C.  It may be so, that the Left Extruder temperatures are (incorrectly) taken from right extruder settings and vice versa. Similar problems occurred with right extruder (see my earlier topic 11 days ago). I haven't done systematic tests, this is just a quick observation.

**LEFT** Extruder:
[image]

**RIGHT** Extruder:
[image]

guillemavila, 2017-12-15:
Can you upload the gcode?
Thanks

S3D-r, 2017-12-18:
Here you are:  

https://www.dropbox.com/s/9cwdcbvgwh6yqy2/Temperature%20problem.gcode?dl=0

closed 2018-06-14 by guillemavila