BCN3D/Cura #48: Cura for Mac 1.1.0 Beta "BCN3D Cura" is damaged and can not be opened. You must eject the disk image.

flandaluce, 2017-12-27:
Intentando instalar la última versión de Cura para Mac
===============
“BCN3D Cura” está dañado y no puede abrirse. Debe expulsar la imagen de disco.
“BCN3D Cura” está en la imagen de disco “BCN3D.Cura-1.1.0-beta.dmg”. Safari descargó esta imagen de disco hoy a las 17:40 desde github.com.
===================
Descargado varias veces alguna idea?
Trying to install the latest version of Cura for Mac
===============
"BCN3D Cura" is damaged and can not be opened. You must eject the disk image.
"BCN3D Cura" is in the disk image "BCN3D.Cura-1.1.0-beta.dmg". Safari downloaded this disk image today at 17:40 from github.com.
===================
Unloaded several times some idea?

eloipr, 2017-12-27:
I have updated the file.
Download it again from the release.

eloipr, 2017-12-28:
Write in the terminal:
`sudo spctl --master-disable`
This will let you install non Apple apps.

flandaluce, 2017-12-28:
Works, thanks

closed 2017-12-28 by flandaluce