BCN3D/Cura #53: M104 S???

cyclemadkiwi, 2018-01-15:
I am printing and wanting a Temp of 240°C the first layer is correct, then in the code at each layer after that I see M104 S248.8. I have no other temp settings visible and even after un-hiding them there is no setting to this number...in fact the M104 S248.8 is sometimes 0.1°C different (up or down) throughout the code.

guillemavila, 2018-01-19:
Do you have `Auto Temperature` switched on?

cyclemadkiwi, 2018-01-19:
I don't use it...never liked it. The Auto Temperature was hidden....I made
it visible and yes it was ticked!! I have unticked it and hidden it again
and the code was good....
Seems you have to check the hidden elements at every update lol

Thanks :-)

Regards
Glen

closed 2018-03-05 by cyclemadkiwi