BCN3D/Cura #54: Grouping issue

cyclemadkiwi, 2018-01-15:
When I select "Multiply Selected Model" and make copies....I group them to move them and I get the group moving to random location on it own. Try to move them manually and when you release the mouse button them move back to where you moved them from!

eloipr, 2018-01-18:
For me it is working fine.
Is this happening when using duplication/mirror mode or regular mode?

cyclemadkiwi, 2018-01-18:
​This was in regular mode! I just tested and it works fine! Maybe I did
something wrong before, my friend Carl also had same issue....I think it
was an error we made. Before I can try it again (The 1.1.0 Beta) the Nozzle
Temp issue will need fixing....keep up the great work :-)

Regards
Glen​

eloipr, 2018-01-19:
Thanks for your feedback, we'll check the Nozzle Temp issue you've mentioned. :D

closed 2018-01-19 by eloipr