BCN3D/Cura #58: Mirror and duplication fail when the right hand hotend is selected as the support extruder before selecting mirror or duplication mode

IdiocyCubed, 2018-02-14:
TLDR:  You need to set the left hand hotend as the support extruder when using duplication and mirror, otherwise T1 is called in the gcode which breaks the duplication and mirror offset.

When printing in duplication or mirror the print fails on the first layer because there is a call to T1.  This occurs when the right hand hotend is selected as the support extruder before changing to duplication and mirror.  Once duplication and mirror is selected, there is no option to change this back without leaving duplication and mirror mode and manually changing the support extruder back to the left hand hot end.  So you will need to remove all calls to T1 when in duplication and mirror (This should be checked for anyway)  and or set the left extruder as the default extruder for everything when duplication or mirror are selected.  @cyclemadkiwi @wanago

eloipr, 2018-02-15:
Which version of BCN3D Cura are you using?

cyclemadkiwi, 2018-02-15:
This is using the latest 1.1.0 official release

eloipr, 2018-02-15:
I've tested it with the Sigmax and it's working. Can you send me the gcode?
When changing to duplication/mirror modes all the support settings are changed to the left extruder. When you go back to regular mode you should see that all these setting are set to the left extruder.

IdiocyCubed, 2018-02-15:
Here is the entire file:
[Lower Door Knob 1.1.txt](https://github.com/BCN3D/Cura/files/1727277/Lower.Door.Knob.1.1.txt)

Here is the gcode header, and the the offending section is below as well

`;FLAVOR:Marlin
;TIME:11807
;Filament used: 2.90396m
;Layer height: 0.15
;DUPLICATION
;BCN3D_FIXES
;Generated with Cura_SteamEngine 1.1.0
M190 S105
M104 S245
M104 T1 S245
M109 S245
M109 T1 S245
;Sigma ProGen 2.1.0 (Build 29BI1651)
;BCN3D Fixes applied
; - Heat only essentials
; - Fix Tool Change Z Hop
; - Fix First Extrusion
; - Fix Temperature Oscilation
; - Cool Down To Zero At End

G21          ;metric values
G90          ;absolute positioning
M82          ;set extruder to absolute mode
M204 S2000   ;set default acceleration
M205 X15 Y15 ;set defaul jerk
M107         ;start with the fan off
G28 X0 Y0    ;move X/Y to min endstops
G28 Z0       ;move Z to min endstops
G1 Z5 F200   ;safety Z axis movement
T0           ;switch to the left extruder
M605 S4      ;clone extruders steps
G92 E0       ;zero the extruded length
G1 E20 F50   ;extrude 20mm of feed stock
G92 E0       ;zero the extruded length
G4 P2000     ;stabilize hotend's pressure
G1 F2400 E-8 ;retract
M605 S3      ;back to independent extruder steps

;LAYER_COUNT:133
M605 S5 ;duplication mode enabled
G4 P1
G4 P2
G4 P3
;LAYER:0
M107
M204 S1750
M205 X5.62 Y5.62
G0 F10212 X76.554 Y286.319 Z0.2
M205 X4.69 Y4.69
;TYPE:SKIRT
G1 F2808 X76.36 Y286.85 E0.00621
G1 X76.092 Y287.348 E0.01241
G1 X75.757 Y287.804 E0.01863
G1 X75.344 Y288.223 E0.02509
G1 X74.88 Y288.629 E0.03185
G1 X74.294 Y289.055 E0.03981
G1 X73.799 Y289.35 E0.04613
G1 X73.294 Y289.604 E0.05234
G1 X72.961 Y289.726 E0.05623
G1 X72.224 Y289.96 E0.06472
G1 X71.567 Y290.11 E0.07212
G1 X66.692 Y290.791 E0.12616
G1 X65.874 Y290.82 E0.13514
G1 X59.743 Y290.413 E0.2026
G1 X54.311 Y290.359 E0.26223`

The offending section is here

`
;TYPE:SKIN
G1 F720 Z0.2
G1 F1200 E27.331
G1 F2808 X51.975 Y241.292 E27.53695
M205 X5.62 Y5.62
G0 F10212 X52.15 Y241.689
M205 X4.69 Y4.69
G1 F2808 X52.15 Y283.776 E27.73909
M205 X5.62 Y5.62
G1 F2100 E22.23909
G1 F720 Z0.5
G0 F10212 X38.15 Y284.351 Z0.5
M205 X4.69 Y4.69
G1 F720 Z0.2
G1 F1200 E27.73909
G1 F2808 X38.15 Y241.111 E27.94677
M205 X5.62 Y5.62
G0 F10212 X37.975 Y241.375
M205 X4.69 Y4.69
G1 F2808 X37.975 Y284.087 E28.15191
M205 X5.62 Y5.62
G0 F10212 X37.8 Y283.56
M205 X4.69 Y4.69
G1 F2808 X37.8 Y241.921 E28.3519
M205 X5.62 Y5.62
G1 F2100 E12.3519
G92 E0

T1    <-------- This is called randomly here, just before the skirt???
G92 E0
G91
G1 F12000 Z5
G90

M109 S245
M104 T0 S150
G0 F12000 X12.541 Y247.921
G0 Z0.5
M205 X4.69 Y4.69
;TYPE:SKIRT
G1 F720 Z0.2
G1 F1200 E0
G1 F2808 X12.114 Y241.105 E0.07497
G1 X12.104 Y240.793 E0.0784
G1 X12.089 Y234.942 E0.14263
G1 X12.102 Y234.567 E0.14675
G1 X12.493 Y228.962 E0.20844
G1 X12.578 Y224.94 E0.2526
G1 X12.541 Y197.913 E0.54931
G1 X12.102 Y190.53 E0.6305
G1 X12.104 Y189.885 E0.63758
G1 X12.162 Y189.374 E0.64323
G1 X13.344 Y182.216 E0.72287
G1 X13.874 Y176.88 E0.78174
G1 X14.103 Y172.191 E0.83328
G1 X14.164 Y171.61 E0.83969
G1 X14.366 Y170.368 E0.85351
G1 X14.509 Y169.74 E0.86058
G1 X14.652 Y169.32 E0.86545
G1 X15.067 Y168.252 E0.87803
G1 X15.337 Y167.666 E0.88511
G1 X15.642 Y167.17 E0.8915
G1 X16.499 Y165.935 E0.90801
G1 X16.898 Y165.43 E0.91507
G1 X17.279 Y165.047 E0.921
G1 X18.113 Y164.291 E0.93336`

guillemavila, 2018-02-15:
Hello, we've tested the same configuration and the gcodes doesn't include any call to T1...
Which OS are you using? so we'll try to reproduce the issue. Does it happen in other computers?

cyclemadkiwi, 2018-02-15:
Win7
I also had the issue that T0 did not print at all because it was at standby
temp during duplication when the print started. Did not save that Gcode.

Have not tried on another computer.

eloipr, 2018-02-16:
Are you using a Sigma? In case you are, what changes have you made to BCN3D Cura to use duplication?

cyclemadkiwi, 2018-02-20:
It is our own home made sigma, we are using 1.2.7 with sigmaX mods. BCN3D
Cura, we are using the SigmaX profile with adjusted bed size

I've not had any issues since that first one. I have done a few mirror and
duplication prints now all fine! I checked out the settings when changing
to Mirror from Regular and it all worked as expected.

Regards
Glen

closed 2018-06-13 by eloipr