BCN3D/Cura #60: Sigmax Start Gcode error

MJFX-645, 2018-02-16:
I noticed an error in the start gcode on the Sigmax

M108 P1      ;enable layer fan for idle extruder

should be

M106 P1      ;enable layer fan for idle extruder

guillemavila, 2018-02-16:
It's not a bug actually. 
M108 is a custom gcode for BCN3D machines which clones layer fan behavior. This way the idle extruder keeps its environment in similar conditions than when printing, resulting in a cleaner tool change operation when this idle extruder comes back to print.

By default layer fan is independent of each extruder, being turned off when switching to another tool.
As our firmware is based on Marlin, M106 parameter is Sxxx only (0-255), to change active toohead's layer fan power. Although RepRapFirmware/Teacup allows M106 Px to give commands to the specific extruder it's not our case.

Very nice appreciation though, thanks for the feedback!

MJFX-645, 2018-02-16:
I only really noticed as I was looking through the gcode to try and sort out why the right extruder wasn’t priming properly and looking up each code to try and find the problem, now I know it wasn’t a typo and actually a nice feature!

guillemavila, 2018-03-06:
In addition, in next Cura 2.0 you'll be able to enable/disable cloning cool fan behavior, in the cooling section.
https://github.com/BCN3D/Cura/releases/tag/v2.0.0-alpha

closed 2018-03-06 by guillemavila