BCN3D/Cura #61: New Feature - Nesting parts

cyclemadkiwi, 2018-02-27:
It would be great if we could nest parts that can fit inside other parts :-)

At the moment, Cura will put a "Bounding Box" around a part, when you place parts on the build plate you may not put parts inside of this area. If you have a large Rectangular part that is hollow right through, you cant use the space in the middle of that part, or an "L" shaped part that you would like to arrange copies in opposite direction to fill the space with only a small gap between. This is a waste and not using the space efficiently.
This would be a GREAT fix/modification to BCN3D Cura!

eloipr, 2018-03-06:
You can do this by unchecking the option "Ensure models are kept apart" in cura preferences.
It's not exactly what you are asking for, but this will let you place the model anywhere in the buildplate.

cyclemadkiwi, 2018-03-06:
That's great, thank you :-)
I think also it is addressed in the Cura 2 Alpha with allowing parts to
"Float" in space. Our SLS printer does this.

Regards
Glen

closed 2018-03-09 by cyclemadkiwi