BCN3D/Cura #62: Cura 2 Alpha Supports

cyclemadkiwi, 2018-03-09:
I have been testing and doing a lot of prints for a production job, I was using ver 1.1 and the supports were great without having to edit them. I tried the new version 2 alpha and saved about 3 hours in print time on the same 19 hour print. That was the good part.....however, something has changed with supports....These new faster print in Ver 2 the supports were a lot more difficult to remove, so much so that I had to report this here!
Not sure if anyone else has found this! I just used the std settings as before....the result was also very ugly for the supported surface that was printed on top of the support.

guillemavila, 2018-03-12:
Not using material support, right? Like pla with pla supports

cyclemadkiwi, 2018-03-12:
Hi Guillem,

Yes that's correct. Using ABS and ABS supports so I can use duplication to
full ability. In Ver 1.1 these supports were easier to remove on the same
print model and same settings.

Regards
Glen

guillemavila, 2018-05-03:
Hopefully fixed, test it with the next release of 2.0.1 :)
https://github.com/BCN3D/BCN3DSigma-Profile-Generator/commit/fdfb61f81ce9872fb8dca1d2556c840dd882b0a2

guillemavila, 2018-05-08:
Here you are https://github.com/BCN3D/Cura/releases/tag/v2.0.1

closed 2018-06-13 by eloipr