BCN3D/Cura #63: BCN3Dcura 2.0.0 alpha: Create material button does nothing

sk4477, 2018-03-09:
I cannot create a new material from scratch. Clicking the create material button does nothing. Duplication works normal. 
System affected: BCN3D_cura 2.0.0 alpha running on Win10. 
[image]

sk4477, 2018-03-16:
This bug is still around in 2.0.0 beta
(and apparently it is also in Cura 3.2.1)

eloipr, 2018-03-16:
When you click the "Create" button, the new material is created under ther "Custom" category. You have to scroll down and you'll see it.

sk4477, 2018-03-16:
I wish.. But it doesn't. No custom material category is created, no new material added. 
See https://github.com/Ultimaker/Cura/issues/3367

But: since it is an UltiMaker Cura issue, i guess i better join the thread there. 
Thanks anyway

guillemavila, 2018-05-08:
Fixed with [v2.0.1](https://github.com/BCN3D/Cura/releases/tag/v2.0.1)?

sk4477, 2018-05-11:
Yes, appears to be fixed. Installed 2.0.1, deleting both appdata folders, problem can no longer be recreated. Is working now as intended.

closed 2018-06-13 by eloipr