BCN3D/Cura #66: Where to find the Vitamin's Settings?

thesec, 2018-03-11:
"Reduce Retraction", "Prevent Filament Grinding", ... seem all gone with 2.0?
Where to find each?

thesec, 2018-03-11:
Especially "Heat Essentials" is a must-have.

guillemavila, 2018-03-12:
Heat Essentials is included by default. Cura only heats up the used extruders. No settings.
Prevent Grinding under Material settings.
Reduce Retraction disabled as it was doing more harm than good... Play with combing, retraction minimum travel and prevent grinding to get the same effect yet better print quality.

closed 2018-03-12 by guillemavila

thesec, 2018-03-12:
Cura does indeed NOT only heat up "essentials": see https://github.com/Ultimaker/Cura/issues/1128 and references.

thesec, 2018-03-12:
Also, just slightly OT but while at it: does the "was ... designed" imply the plugin being discontinued now?

guillemavila, 2018-03-12:
In Cura 2.0 we never had this issue. Note https://github.com/Ultimaker/Cura/issues/1128 dates from 2017, before UM3.2, which our 2.0 is based on. That's why I closed the issue. Cura heats up only the extruders it's going to use.

Yes, Vitamins plugin is discontinued. We put our effort into integrate our features in the most reliable way. That is native. This doesn't mean that maybe in the future we test something straight in the Ultimaker's version, and as we are open source I usually share it as the plugin so the community can benefit and we can get feedback from you. But developing the plugin parallel to our version is not in our roadmap.

thesec, 2018-03-12:
Thank you for the confirmation.