BCN3D/Cura #67: Model Positions not Saved with Project

thesec, 2018-03-11:
Load a model
re-position it, e.g. at the upper right of the bed
save the project
clear the plate
load the saved project
find the model positioned automatically at bed center

expected: find the model at the saved position

thesec, 2018-03-11:
Skirt Extruder, Prime Tower, there's plenty other as well that is not saved ... please fix.

thesec, 2018-04-11:
Having to reposition all models in projects gets quite annoying. Is this on the roadmap yet? When?

S3D-r, 2019-03-21:
I was about to ask this same but found your post. I can see that your issue hasn't gained a lot of attention... 

Not sure about the model positioning, but all the customized settings appear NOT to be transferred by project saving and re-opening. Current version is 2.1.5.

S3D-r, 2019-06-03:
This is 1 year and 3 months old issue, yet no-one from the development team has responded. The issue is a top priority: currently, you can't reliably save your settings once you've found the good ones. Well, maybe "Save new Profile" is some level of go around, but a final solution to the clear bug should be done.

S3D-r, 2020-02-27:
This bug appears to be still present in current version 3.1.0.  Two years after reporting...