BCN3D/Cura #74: Not enough material ready

maodch, 2018-03-17:
In the latest version it's impossible to print something, because at the begining of each cycle (skirt, part, tower what ever), the filament comes very late and with to few dense out the nozzle.

With Sigma it's unusable. Did you change something with the feedrade (maybe took the one for sigmax (bondtech) instead the sigma (non bondtech).

I downgraded the version to 1.1 (BCN edition) and also tired the original cura fom Ultimaker without having this problems.

guillemavila, 2018-03-19:
Hmm.. strange, as coast/wipe settings are the same in 2.0 and 1.1
Do you have the failing gcodes so we can check them?

guillemavila, 2018-04-12:
Same issue as https://github.com/BCN3D/Cura/issues/85 ?

closed 2018-04-20 by guillemavila