BCN3D/Cura #75: Wrong extruder prints the initial layer

S3D-r, 2018-03-21:
**Settings**:
Object extruder: Left 
Support (PVA) extruder: Right 
Built Plate Adhesion: Skirt
Built Plate Adhesion Extruder: Left

The object has flat surface against build plate, i.e. there should be no support below it. Yet the Cura makes the right extruder (=PVA support) to print the first layer, which appears to be slightly larger than object's bottom surface (i.e. makes kind of brim).

S3D-r, 2018-03-21:
Here u r:
Code: https://www.dropbox.com/s/26kbrbhqvinb9na/First%20layer%20issue.gcode?dl=0
STL: https://www.dropbox.com/s/z9gb42ha8tktsms/First%20layer%20issue.stl?dl=0

guillemavila, 2018-03-22:
Looks like the part is lifted 0.x mm over the buildplate, and then its all printed on a support base. For me the stl slices properly, starting with model material and pva.
[image]

Do you have `Automatically drop models to the buildplate` under Preferences > Configure Cura?

S3D-r, 2018-03-22:
That's what I suspected also. But I have that  "Automatically drop models to the build plate" option chosen, also z position shows 0. I tested again today, and same issue happens again...
See video: https://youtu.be/igHccN0C2ME

guillemavila, 2018-03-22:
Thanks, the video has been very useful! 

Sorry I was testing with the public beta of the upcoming Cura 2.0.
It's a 1.1 known issue, if you wanna test the same part with the 2.0 beta you'll see the problem has been solved. (Both 1.1 and 2.0beta can be installed in parallel)

Sorry for the inconvenience, we'll release the final version very soon.

S3D-r, 2018-03-23:
You should not assume that the users have the latest _beta_ in their use.

You recommend Cura 2.0b, but did not give any hint where to download it. Can't find it anywhere.

guillemavila, 2018-03-23:
Yep, my mistake ;)

https://github.com/BCN3D/Cura/releases

guillemavila, 2018-04-23:
Working from v2.0+

closed 2018-04-23 by guillemavila