BCN3D/Cura #81: Ver 2.0 Not importing or creating Profiles

cyclemadkiwi, 2018-04-05:
Hi,
This has still not been addressed!!!!!

This was an issue way back in Ver1.03 and you fixed it.....in the Ver 2 Alpha I pointed out the problem of creating and importing profiles....THIS IS STILL NOT WORKING

I have installed the latest Ver 2.0 and I went "Import Profile" and it said success but it did not show in the list. restart prog...still not there...restart PC....still not there.

Please fix this, only I can use the printer, my fellow workers can't use as they do not know the settings.

eloipr, 2018-04-06:
Thanks for reporting. We've noticed that the profile management with Sigmax is not working. We'll make a release soon fixing it.

guillemavila, 2018-05-03:
Sigma/Sigmax old profiles import fixed https://github.com/BCN3D/Cura/commit/0092856e025dfeeed04429446d6a6ca5a27a361f

closed 2018-05-03 by guillemavila