BCN3D/Cura #83: Ver. 2.0 Custom Profiles

MJFX-645, 2018-04-09:
I can’t seem to import my custom profiles from Ver. 1.1.0, Ver. 2.0 says they have been imported successfully but I can see them in the mange profiles selection?

Also I can’t create a new profile from the built in profiles, every time I try it tries to create one with an incrementing number e.g. "Draft Quality #3" etc. but It doesn’t show in the manage profile window?

thesec, 2018-04-11:
Changing the setting_version to 4 in the according JSONs should get them to show.

MJFX-645, 2018-04-11:
Thanks for your comment could you tell which JSON file(s) to modify, i had a quick look and can't find any files called "according"

thesec, 2018-04-11:
Look for *.cfg, e.g. in the 'quality' folder within your settings directory. Aside the 'setting_version' you might also want to check the 'version' (should imo be 3). You can compare the values to their equivalents in the stock profiles within the application's dir.

MJFX-645, 2018-04-11:
I have looked in the quality folder and I can't find my profiles in there just the built in ones, I have tried importing them again and they haven’t appeared.

I have reinstalled Ver. 1.1.0 to look for my profiles, they appear in the slicer, in that folder and can't find them either?

thesec, 2018-04-11:
Hmmm, what do you refer to as "importing"? Have you tried to move them directly (btw - try with one first and make sure to include the configs it references to)? Which folder are you looking at and in which OS?

MJFX-645, 2018-04-11:
Im using windows 10, importing I did via the profile manager within cura in the preferences. 

The folder i was looking in was C:\Program Files\BCN3D Cura 2.0\resources\quality\bcn3dsigma

I did try and drag and drop the exported(from ver 1.1.0) into this folder but they are different file formats

thesec, 2018-04-11:
Never used the profile manager for imports ... :/
But the file formats should be the same, both versions use plain text files ending in .cfg ...

Also the folder your profiles should go in is the user's folder (should be users/username/appdata/... in Windows) - you save yourself from re-applying changes after program updates when you keep yours in there and is this the location where Cura keeps its history and your personal profiles anyway.

MJFX-645, 2018-04-11:
Still not showing after dropping all the cfg files into the correct file and changing the setting_version to 4 and the version to 3. If I try and duplicate the built in profiles within cura the new profile appears in the folder but not in cura, so I’m in the correct folder they just won’t show up in cura!

thesec, 2018-04-11:
:/
The described proceeding solved it for me when migrating profiles.

You could diff a single profile of yours vs the stock version of it in 2.0 to see which other differences might result in the behaviour you experience ... or wait for one of the devs to chime in ...

eloipr, 2018-04-12:
Hey! @MJFX-645 what you are doing seems to be correct, but the version should be 2. The setting_version it's okay with 4.

thesec, 2018-04-12:
The version should be the same as with the Ultimaker Cura version, at least this would be very nice to have.
I hope that was not changed as well in the BCN3D fork?

MJFX-645, 2018-04-12:
I have tried with the version set to 2 and the setting_version set to 4 and they still dont show

guillemavila, 2018-04-20:
Same as #83

closed 2018-04-20 by guillemavila

thesec, 2018-04-20:
Same? It is #83?!

guillemavila, 2018-04-20:
hahah lol. Sorry same as https://github.com/BCN3D/Cura/issues/81

MJFX-645, 2018-04-20:
Proberbly a really simple answer, but how do I get the fix? The windows installer hasn’t been updated as far as I can see?

guillemavila, 2018-04-20:
It's fixed in the source code. You can compile your own version or wait until we publish the 2.0.1 soon with some bugs fixed (including this).
Thanks for your patience and sorry for the inconvenience