BCN3D/Cura #93: 2.0.0

DarrylUK, 2018-05-02:
**Application Version**
2.0.0

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Select Black ABS, any quality and infill.

**Actual Results**
Temperature on extruder 1 is set at 0°C.

**Expected results**
Temperature on extruder 1 should be 250°C.

**Additional Information**

DarrylUK, 2018-05-04:
We are printing with the left one (0) but neither heat up. As you begin the print, if you go into settings (to view the target temperature), both extruders are 0.

guillemavila, 2018-05-04:
Hmm for me the temperatures are ok, could you send the gcode (or at least the first 100 lines)?

On the other hand, couldn't it be it's still heating the bed? It won't give the command to heat the Hotends until the bed has reached its target.