BCN3D/Cura #94: Profiles disappear after update

v-iarosenco, 2018-05-10:
After upgrading to v.2.0.1 (which is actually uninstalling v.2 and installing the new version) created profiles disappear. I didn't have any created materials to check them as well...

guillemavila, 2018-05-10:
Hello,

First of all no panic as your custom profiles are saved and you'll be able to recover them.
This is a normal behavior. Uninstalling v2.0.0 is an option the installer gives you when upgrading to 2.0.1 (I understand you're working on Windows?).
With this update, profiles change and it might be cause of conflict, so it detaches from profiles from older versions. However, you can recover them by:

- Option 1: You reinstall v2.0.0, the profiles will be there, and you'll be able to export them and then import to v2.0.1
- Option 2: With cura v2.0.1 closed copy the folders `C:\Users\[your user]\AppData\Roaming\BCN3D Cura\2.0.0\quality` and `C:\Users\[your user]\AppData\Roaming\BCN3D Cura\2.0.0\materials` to `C:\Users\[your user]\AppData\Roaming\BCN3D Cura\2.0.1\quality` and `C:\Users\[your user]\AppData\Roaming\BCN3D Cura\2.0.1\materials`

v-iarosenco, 2018-05-10:
Thank you, yes, I use Windows 10.
It worked well, is there something else to copy for settings visibility? (i made more settings visible in v.2)

guillemavila, 2018-05-10:
Awesome!
To copy the settings visibility... It's a bit more work but you can also edit the file `C:\Users\[your user]\AppData\Roaming\BCN3D Cura\2.0.0\BCN3D Cura.cfg` and copy the line `visible_settings` under `[general]` into `C:\Users\[your user]\AppData\Roaming\BCN3D Cura\2.0.1\BCN3D Cura.cfg`

v-iarosenco, 2018-05-10:
Thanks a lot, all works.

closed 2018-05-10 by eloipr