BCN3D/Cura #96: settings unreadable

koen330, 2018-05-14:
**Application Version**
Cura 2.0.1
Mac OS

Most of the settings become unreadable on MacOS. Happens on different macs.

[image]

MarkRAndrews, 2018-05-21:
I have the same problem on my Mac...

Krammer002, 2018-05-26:
Update:
I have just installed the Ultimaker Cura 3.3.1 on to the same machine as the latest BCN3D Cura 2.0.1...
 Cura 3.3.1 displays fine, Cura 2.0.1 has the font issues...

jmh68300, 2018-05-28:
I try the same process as Krammer002, I confirm that BCN3D Cura 2.0.1 has the fonts issues on MAC OS 10.13.4. This is a known problem, simple to fix during the compilation.

Krammer002, 2018-05-28:
Great!   Please advise the process to fix BCN3D Cura 2.0.1?

Krammer002, 2018-05-28:
I have not compiled this app before. Are there instructions available how to do this?

jmh68300, 2018-05-28:
you have to use cmake. You can see https://github.com/Ultimaker/cura-build
I don't have all requirements for compiling this application (I don't have the 3.5 version of python and I don't want to change my setup...

Krammer002, 2018-05-28:
I don't have the time to devote this at the moment, and I'm unsure why the devs cant address this issue as its so simple...!

guillemavila, 2018-06-14:
We will really appreciate your feedback if you can test this alpha

https://goo.gl/ixmywp

Krammer002, 2018-06-14:
Hi
With pleasure... won’t be until the weekend though.
Cheers

Mark Andrews

Old Orchard 
St. Cyors 
Luxulyan 
Bodmin 
PL30 5EA

( Direct:	+44[0] 1726 852161   
( Mobile:	+44[0] 7597 621530  
* Email:		mark@ndrews.net

jmh68300, 2018-06-14:
Hi,

I've just tested it and the fonts look good for me. I didn't test anything else. See below:
[image]

But I 've had only one issue and only one time! See below:
[image]

Krammer002, 2018-06-16:
Hi
Fonts look good on my Mac also... apart from:
[image]