BCN3D/Cura #100: Gcode code for pause protocol in Sigma BCN3D printer

Cosomo, 2018-05-25:
Hi everybody,

I'm trying to create a pause at certain layer in a Sigma BCN3D printer. I'm not talking about start manually a pause pressing the touch display, wich of course works well but is not precise.

I have seen that for other printers the command is M0 or M25, but this command is not working for my printer.

Anybody knows what is the right one?

I use the last version from Ultimaker's site, but in this case is not relevant.

Thanks in advance

guillemavila, 2018-05-25:
hahah working on it right now!

guillemavila, 2018-05-25:
Here you are 33c96ef978fdaef78a43d93aebb9bbedfc155232

Cosomo, 2018-05-27:
First of all many many thanks

I have no experience with python. I'm using last version of Cura, 3.3 from Ultimaker's site.
Maybe because of this I've been unable to crete the plugin.

Fortunately I usually use Simplify 3D and with your info I have added this post porcessing command and IT WORKS perfecty!!! (of course, I never doubted it):

{REPLACE "\n; layer X" "\nG69 ;park active hotend\nM25 ;stop SD reading\nG4 P1\nG4 P2\nG4 P3\n; layer X"}

Substitute layer X with the layer BEFORE you want to stop.

Kind regards!

closed 2018-05-27 by guillemavila