BCN3D/Cura #112: Duplication mode right extruder temperature wrong

ianewby, 2018-10-04:
**Application Version** 
2.1.1

**Platform**
Windows

**Display Driver**
NVIDIA

**Steps to Reproduce**
Configured duplication mode on a Sigma, after upgrading firmware and LCD drivers to latest version. 
The resultant GCode does not set the right extruder temperature, resulting in the print failing. Note that on my printer, the right extruder is higher than the left so gets the shim layer added.

;FLAVOR:Marlin
;TIME:49461
;Filament used: 15.9242m
;Layer height: 0.25
;Extruders used: T0 0.5 T1 0.5
;BCN3D_FIXES
;Generated with Cura_SteamEngine 2.1.1
T0
M190 S80
M104 S280
M109 S280
M82 ;absolute extrusion mode
;Sigma ProGen 2.2.0 (Build 26JI1246)
;BCN3D Fixes applied
; - Fix start GCode
; - Fix Acceleration/Jerk commands
; - Prevent Filament Grinding

G21          ;metric values
G90          ;absolute positioning
M204 S2000 ;set default acceleration
M205 X15 Y15 ;set default jerk
M107         ;start with the fan off
G28 X0 Y0    ;move X/Y to min endstops
G28 Z0       ;move Z to min endstops
G1 Z5 F200   ;safety Z axis movement
T1           ;switch to the right extruder
G92 E0       ;zero the extruded length
G1 F73.8 E20 ;extrude 20mm of feed stock
G92 E0       
G4 P2000     ;stabilize hotend's pressure
T0           ;switch to the left extruder
G92 E0       ;zero the extruded length
G1 F73.8 E20 ;extrude 20mm of feed stock
G92 E0
G4 P2000     ;stabilize hotend's pressure

M605 S5      ;enable duplication mode
G4 P1
G4 P2
G4 P3

eloipr, 2018-10-08:
I've tried to reproduced the error but it's working fine for me. Which material are you using?

ianewby, 2018-10-08:
We are using polymaker PC-MAX, with a temperature of 280.

I'll do some more testing - we have switched to mirror mode since having the issue.

ianewby, 2018-10-08:
I've tried again with mirror mode - the left extruder goes to temperature and the right stays at 0. 

Firmware version 01-2.0.3
BCN3d Cura version 2.1.1

Is the gcode file supposed to have the code to set the temperature, or is the temperature set by enabling mirror/duplication mode?

cyclemadkiwi, 2018-11-30:
This error is also occurring for me, I have just now done my first Dup print with the latest version and the right hotend failed to heat. I had to insert code from a known dup file from the old version.

eloipr, 2018-11-30:
I've tried different versions of cura and for me it's working fine with PLA. Also with different hotends.
The gcode I get:

```
T0
M190 S65
M104 S220
M104 T1 S220 ;Fixed T1 temperature
M109 S220
M109 T1 S220 ;Fixed T1 temperature
M82 ;absolute extrusion mode
;Sigma ProGen 2.2.0 (Build 26JI1246)
;BCN3D Fixes applied
; - Fix start GCode
; - Fix Acceleration/Jerk commands
; - Prevent Filament Grinding
```
Which material are you using? Which settings are you changing?

cyclemadkiwi, 2018-12-02:
Hi,

Thanks for the reply.

I have only had it happen the once! I seem to remember it doing this at an
earlier version as well. My next prints using duplication have been fine.

My more important issue that has not had a reply or been addressed is doing
"Updates" and deleting all my custom profiles and materials. Other programs
allow you the option of "Upgrade" or "New Install" I hope this gets some
attention.

Thanks
Glen

cyclemadkiwi, 2019-04-16:
In the New Version 2.1.5 I have had this issue a few times now, mirror or Duplication will not heat the right hotend. I restarted the program, same result, I restarted the PC, same result.
The code to turn the right hotend was missing. It's a random thing too which makes this hard to fault find!
As of today, it is working ok. But from time to time, I have to manually enter the code to heat the right hotend up!