BCN3D/Cura #114: Purge sequence wrong when printing compensation layer for right nozzle in duplication mode

Prejkox, 2018-10-08:
**Application Version**
2.1.1

**Platform**
Win10-64

**Display Driver**
Nvidia

**Steps to Reproduce**
Duplication mode with right nozzle 0.025mm higher than the left. Printed part is "vertical_bowden_v21.55". Material used is Ngen with pre configured profile by BCN3D. Original R17 printer. Printed from SD-card.

**Actual Results**
Printer ask first if I want to print even due to I have no shim. I choose yes. It begins to purge the right nozzle then stop purging. Then it purges the left nozzle and stops. Mean time the left is purging the Right nozzle is loosing pressure and spews out the filament. Now it start to print with the Right nozzle, the extra layer that is needed to compensate for the 0.025 difference. But there is no pressure left so it only is small blobs that is the result of this layer. When the nozzles are primed again it is to less for the right nozzle due to the earlier pressure drop so it is not printing well and is under extruding.  
Result is that Right model does not print good and getting lose due to bad adhesion to bed.

**Expected results**
That Both models are printed good. Also that the nozzle that prints the compensation layer is purged and primed just before it shall begin print. Also that BOTH nozzles are primed proper again before they both begin print simultaneously.   

**Additional Information**
I guess that this is also an issue in mirror mode but I have not tried that.
Maybe this is a firmware related issue. I do not know. Maybe the compensation layer is generated by the printer itself and purge sequence is outside the gcode. I guess you know this ;)

Prejkox, 2018-10-08:
The printer have the latest firmware that was released at the time I wrote this.

AlejandroGarcia92, 2018-10-15:
What you describe at gcode purge sequence, the skirt should be enough until the printed part starts but maybe "vertical_bowden_v21.55" makes the default skip not enough.

Related with FW, the threshold Z offset limit until a dupli/mirror print job both heads don't start together is 0.05mm. You are telling me that you have 0.025mm, so I will test it again.