BCN3D/Cura #115: Wrongly calculated "doesn't fit into the build volume" if using brim.

S3D-r, 2018-10-16:
I have an item which max dimensions (X/Y) are exactly the size of the build table, but the footprint is much smaller than the table. I.e. the object widens as the Z gets bigger. So, there would be a plenty of space for a brim. However, if I add even a small brim, Cura immediately claims the item doesn't fit.

eloipr, 2018-10-17:
This happens because Cura calculates the space based on the bounding box of the model. What you are saying would be a nice improvement!

S3D-r, 2018-11-04:
In my case, the bounding box is the same with the brim and without the brim. So, if Cura first calculates the bounding box without the brim and then just enlarges the bounding box by adding 2xbrim width both to X and Y dimensions, I'd consider this a **clear bug**, not just enhancement.  Cura's claim "doesn't fit into the build volume" is false.

S3D-r, 2019-09-11:
I reported this almost 11 months ago. The latest Cura version (2.1.5) still has this bug...