BCN3D/Cura #118: Cura wastes space in Duplication mode

S3D-r, 2018-11-04:
**Application Version**
2.1.1 and 2.1.2

Printer: Sigma R19.

Duplication mode: I try to pack the table as full as possible. However, Cura doesn't let you place items right to the X-middle line of the table, but it leaves 10...12 mm gap to the center (that is **after** brim). In reality, the items could be packed side to side without a gap (well, you might want some 0,5mm clearance just to keep the parts away).  

That 10...12 mm gap is totally unnecessary and makes sometimes the difference if your item fits into the half of the table or not. 
[image]
[image]

LucasHarbauer, 2018-11-06:
Hello S3D-r,

there is no space wasted in duplication mode. The gap is necessary because of the printer heads. With zero gap there would be a collision between heads.

S3D-r, 2018-11-06:
Hi!

You are talking about the needed gap in Mirror mode, I'm talking about Dublication mode. In Duplication mode print heads are always separated ~100mm from each other, no collision is possible.

Simplify3D does not leave a gap in Duplication mode, which gives ~10% more space in X-direction. In many cases, that's not meaningful, but in some cases, you can load 10...100% more items on the table and in rare cases that 10% makes difference if your item is printable in Dupl. mode at all.

eloipr, 2018-11-07:
Thank you for your feedback. It has been fixed and will be available in the next release!

closed 2018-11-08 by eloipr