BCN3D/Cura #121: Cura freezes it you multiply and group models

S3D-r, 2018-11-08:
Cura gets extremely slow and eventually jams totally if you multiply and group items and try to relocate these groups. Tested with several (4) PC's.

- I place an item. A small, only 84 triangles! 
- I multiply it to 8 copies. 
- I nest (move positions) and make a group of these 8 items.  Making the group takes time, say 15 seconds...1 minute (depending on PC).
- I try to relocate this group. A bit slow in Regular mode, very slow in Dupl. mode.
- I make copies of this group, say 5 copies (=40 items on the table at the same time).  The copies take 30 sec. to appear, and next 30 sec after that you watch the busy-symbol.
- Now, if you try to move one of these groups, Cura jams ("not responding") for several minutes! Even 5...20 minutes. So, practically you are not able to nest these groups on the table. Each moving trial takes minutes or tens of minutes.  (at least in Dupl. mode, not made all possible test in Reg/Mirr/Dubl modes)

It appears to be something fundamentally wrong with Cura how it handles multiple items. No matter what the total triangle count is.

 
-->

**Application Version**

2.1.2 and in quite many earlier versions.

**Platform**

Win10-64

**Steps to Reproduce**

Place an item. Multiplay it (say 8 copies). Make a group of these 8 items. Multiply this group. Try to relocate the copied groups.

**Actual Results**

Total freezing.

**Expected results**

Somewhat fluent operation with no major delays.

**Additional Information**

Problems are worse in Duplication mode.

S3D-r, 2018-12-03:
No comments so far. You other Cura users: do you have this same problem?  
For me, Cura is unusable for slicing multiple items (more than 3 items), I need to use another slicer.

HugoPayens, 2019-03-29:
The same issue, i want slice multiply little objects. Cura freezing

S3D-r, 2019-04-03:
Considering the severity of the issue, I'm a little bit surprised how little this topic has gained attention. No actions/comments from the development team etc...

eloipr, 2019-04-03:
You're right, the multiply and relocation algorithm need to be improved. We are trying to fix it for the next releases.

HugoPayens, 2019-04-03:
Good to hear..  Thank you for the reply!!

Mit freundliche Grüßen
Ronny Köhler-Terz

Prejkox, 2019-08-01:
I also have this issue in current release 2.1.5. It is so slow so it is almost impossible to use.

S3D-r, 2020-02-26:
This appears to exist in the latest version (3.0.1), too.  I have high expectations for the major release (2.x --> 3.x), but no progress what comes to this fault.

conraddewet, 2020-02-26:
Same problem. Horribly slow in duplication/move  mode as well. The same models works 100% with S3d I use for my other printers.