BCN3D/Cura #124: BCN3D Cura New Releases deletes profiles

cyclemadkiwi, 2018-11-22:
Hi,

Every time I do an "Update" it is a totally new install and all the print/material profiles I have worked on all vanish. I have to export all of them in the old version and one by one import them to the new version of Cura. Sometimes this does not work! Why cannot each new version be an update instead of complete new install.?

Each new "Update" gives me a big headache and I loose a lot of time...... Lost time = Lost money!!!

Thank you for using Cura!
-->

**Application Version**
2.0.2

**Platform**
Windows 7 Pro

v-iarosenco, 2018-12-06:
Check issue #94 (in closed), BCN advised how to copy profiles, materials and configurations settings from one version to another. 
However I agree with you completely.

cyclemadkiwi, 2018-12-06:
Thank you, I have made a note of this for future updates. :-)

S3D-r, 2019-03-21:
What is said in issue #94, is actually getting around of the problem, not solving it. Almost any other software in the world does keep the settings in upgrades - no complicated tricks are needed. That's the standard Cura should also follow.

cyclemadkiwi, 2019-03-21:
I Agree with S3D-r there should be an "update" option not a full install. Even after using the export/Import option, the imported profiles appear in the Roaming appData folder for the new version, however, the profile does not appear in Cura. You import again, it says "Success" and it creates a ver2 of your profile in the folder but still doesn't appear in Cura!

cyclemadkiwi, 2019-04-16:
When I "Import" or copy profiles using the method to "Work around the issue" some of them tell me success, then they are hidden.....this is very frustrating. I end up having to "Re-Create" the profile from the beginning.
Please fix the new versions to update the program like other software does.......not a new install!

thesec, 2019-04-16:
Just happened with an update from 2.1.4 to 2.1.5 ... you are making us loose our work and waste our time. Please fix this annoying issue, thank you.

S3D-r, 2020-02-08:
Well, we have now version 3.1.0 available. No improvement in this issue.