BCN3D/Cura #125: Part weight displaying as 0

v-iarosenco, 2018-11-29:
BCNCura 2.1.1
Windows 10 x64
Firmware v01-2.0.4

Weight (mass) of the part on printer's screen is displayed as 0 presumably for 3 digit numbers. For 1 and 2 digit weights it's displayed correctly.

v-iarosenco, 2018-11-29:
Printer - Sigma original 2016

AlejandroGarcia92, 2018-11-29:
Can you share us the header's gcode or full gcode? btw

v-iarosenco, 2018-11-29:
Hi Alejandro,
I checked g-codes and tried some more ideas. The issue is not with three digit weight as I thought, but with our right extruder (0.8 mm) - length of filament printing with it is 0 m; g-codes from the left extruder (0.4 mm) are correct.

Please see the g-code attached and the header below.

;FLAVOR:Marlin
;TIME:22065
;Filament used: 0m
;Layer height: 0.2
;Extruders used: T1 0.8
;BCN3D_FIXES
;Generated with Cura_SteamEngine 2.1.1
T1
M190 S70
M104 T1 S200
M109 T1 S200
M82 ;absolute extrusion mode
;Sigma ProGen 2.2.0 (Build 26JI1246)
;BCN3D Fixes applied
; - Fix Acceleration/Jerk commands
; - Prevent Filament Grinding

G21          ;metric values
G90          ;absolute positioning
M204 S2000 ;set default acceleration
M205 X15 Y15 ;set default jerk
M107         ;start with the fan off
G28 X0 Y0    ;move X/Y to min endstops
G28 Z0       ;move Z to min endstops
G1 Z5 F200   ;safety Z axis movement
T1           ;switch to the right extruder
G92 E0       ;zero the extruded length
G1 F189.0 E20 ;extrude 20mm of feed stock
G92 E0       
G4 P2000     ;stabilize hotend's pressure
T0           ;switch to the left extruder
G92 E0       ;zero the extruded length

G92 E0
G4 P2000     ;stabilize hotend's pressure

[S_RD7-30-1-1.zip](https://github.com/BCN3D/Cura/files/2629403/S_RD7-30-1-1.zip)