BCN3D/Cura #126: Bed preheat temperature

v-iarosenco, 2018-12-06:
Firmware v01-2.0.3

Bed preheat temperature is currently set either to the highest from two materials or to the latest loaded (say you have PLA on one side and ABS on the other side). It's not convenient, if you choose to print with PLA, but the bed preheats to 90 degrees, because the second spool in the printer is ABS.
It would be great, if you could set preheat temperature.

AlejandroGarcia92, 2018-12-10:
There is no plans to implement this feature. Btw, you can set the load temperatures by using Custom Fil.

v-iarosenco, 2018-12-10:
Ok, I can give you a recent real life example - I loaded ABS on the right, then PLA on the left. So now bed preheat temperature is 90 no matter which extruder I use.
But then I unloaded ABS and the temperature is still 90.
I know about custom filaments settings, but it doesn't look like a real solution, does it?

AlejandroGarcia92, 2018-12-10:
I mean that if you load PLA at the left and Custom at the right with a lower bed temp, the bed target temp will change. It is trick, not a solution.

v-iarosenco, 2018-12-10:
...and when I print next with ABS I can't preheat the bed to 90 degrees..
I understand it's not a serious issue, just wanted to point it out to you, don't worry.

AlejandroGarcia92, 2018-12-10:
Ok, I've just tag the issue as a enhancement proposal.