BCN3D/Cura #127: Hotend temperature af

v-iarosenco, 2018-12-06:
Firmware v01-2.0.3

After loading/unloading filament the hotend remains heated. 
It would be great if heating turns off after some time - 1-2 minutes.

AlejandroGarcia92, 2018-12-10:
This time is 10min.

v-iarosenco, 2018-12-10:
I think less time makes more sense - if you loaded filament to start printing straight away, then 1 minute will suffice, if you loaded to print later, then again you don't need to keep it heated for too long. What do you reckon?

Prejkox, 2019-05-12:
I did open the same issue/question in the firmware section some time ago. It is still open.
One more reason to update this (did not mention this in my open issue) is that there are filaments that get burned when no new filament is pushed thru like PVA and clogs the nozzle. What i have experienced is that when the hotend stays on (at printing temperature that is the case here) and do nothing, is that a lot of the filament inside just pours out of the hotend. The risk is that whats left inside on the walls of PVA gets burned and can't be melted again even if reheated. I have had some issues with this. Its really hard to get that nozzle/hotend clean. I use a lot of pva both home and at work.