BCN3D/Cura #132: Gradual infill not working properly, printing in the air etc...

S3D-r, 2019-03-20:
**Application Version**
2.1.5

**Platform**
Win 10-64

I make gradual infill, 3 steps. Cura appears not to print all the steps but just the first (12,5% x 15%) and last (100% x 15%) step. And what is more crucial - it doesn't bridge the denser infill on top of the previous step but prints in the air! Everybody knows this infill will fail in real life. Printing in the air happens also when I use 1 step only.

Video: https://youtu.be/uLb9kNho9mE

S3D-r, 2019-05-27:
No reactions to this so far. My opinion is, that gradual infill does not work at all, nice idea but implementation has failed.

mteterina, 2019-06-19:
I am sorry for the late reply. 
I have looked in to your issue and done some tests myself. 
The gradual infill did work out for me. I printed both square shapes and models that have rounded shapes and overhangs. I understand your concern due to the infill printing almost in the air.  Nevertheless the infill is printed with slight angle if the model is rounded which makes it possible to print, it may be a bit messier than regular infill but printable. I attached a picture where i opened up one of the prints where i used gradual infill to show how it looks for better understanding. 
I also noticed in you YouTube video that you are using only 2 walls? If you are going to print with gradual infill i would recommend you to print with the default settings for the wall line count (4 walls).
If you want you can send me your stl file that you want to print and i can test print it for you and see if it works. 
I hope this answered some of your concerns.

Kind regards 
Maria
 
[image]