BCN3D/Cura #133: Duplication model heights

cyclemadkiwi, 2019-03-21:
Duplication or Mirror Mode
When I have 2 components with different overall heights, sometimes the print will finish at the height of the lowest height part. This includes putting a "roof" on the part that should be taller. 

This has happened in the last 2 or 3 versions.

**Application Version**
2.1.5

**Platform**
Win7 Pro

cyclemadkiwi, 2019-03-21:
[image]

cyclemadkiwi, 2019-03-21:
You can see the shadow of the full height, but the slicing only goes as far as the lowest part.
I also changed to "Regular" Mode and it did the same thing!

eloipr, 2019-04-02:
I can't reproduce this error.
Which printer are you using Sigmax R19, Sigma R19, Sigma or Sigmax?
Wich hotend and profile?
Wich settings are you changing?

cyclemadkiwi, 2019-04-16:
This is on our own homemade Sigma R17, I think I have found out something.
If you first put a short part on the build plate, then a tall part, when you slice it, it will only slice to the level of the short part. If you place the tall part on the build plate first, it will slice to the correct height.

When you slice it, and you go to Layer View, you can see this!

cyclemadkiwi, 2019-04-16:
Hi,

I replied on the Github thread to this issue....I lost this email, but now
I found it!
I hope it helps to understand the problem. Thanks.
If you did not get it, please let me know and I'l write it again.

Regards
Glen Lewis