BCN3D/Cura #134: "Enable prime tower" is on as default - should not be.

S3D-r, 2019-03-25:
Sigma/x never needs the prime tower, as it purges filament into the bin when tool change.  Yet this option is not only visible but it's even "on" by default. No big deal, but clearly something that should be changed. At least "off" by default and preferably non-visible. Maybe even completely removed from settings.

eloipr, 2019-03-26:
We set the prime tower setting to improve the printing quality. Without it we don't get the quality we want to reach. But depending on the geometry of the part you can print it without prime tower.

S3D-r, 2019-03-26:
OK, I see. Why does the prime tower offer better print quality than purging (and rubber blade wiping)?

eloipr, 2019-03-26:
Because the prime tower stabilizes the material flow inside the hotend

S3D-r, 2019-03-26:
I thought the purging was supposed to do the same. But it doesn't ???

mteterina, 2019-06-19:
The prime tower is only used for dual prints, with a tower you reduces oozing, under extrusion, colour contamination and enhances overall printing quality. We really recommend to use it when you are printing with both extrudes. It is not necessary for single extrusion prints and the tower is disabled if you only print with one extruder. If you are concerned about the extra time and material required to print the tower i would recommend you to make it smaller (prime tower size and prime tower thickness), if your print is tall i would recommend to  keep the tower as it is, due to higher risk for the tower to fail/fall if its to thin. 

Kind regards
Maria