BCN3D/Cura #135: Duplicate Mode does not work

kevin-pumpstudios, 2019-04-09:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
 Add the steps needed that lead up to the issue (replace this text) I load the part in duplication mode and set everything up, save to the SD card load it on the printer and start the file 

**Actual Results**

**Expected results**

**Additional Information**

eloipr, 2019-04-12:
Could you give us more information in order to reproduce the error?
Material, Hotend, Profile, changed settings...

kevin-pumpstudios, 2019-04-12:
Both nozzles are loaded with PLA, I have sueccesfully printed with
multi-material (PLA and PVA)
[image: image.png]
When I run the duplicate mode print here is what is on the screen and
printer
[image: image.png]
I have attached the CURA profile as well.  (Cannot think of any additional
information I can provide)

Thanks

Kevin Keller

Pump Studios
1023 Springdale rd - Bldg 4A
Austin TX 78721

512 628 0921 p
512 628 0923 f
512 587 6776 c

cyclemadkiwi, 2019-04-16:
I have the same issue!
It is random, I have to restart Cura and or restart the PC a few times to make it work again!