BCN3D/Cura #138: No scaling of large models

thesec, 2019-05-06:
v2.1.5 does not scale models in meters / thousands of milimeters at all.

mteterina, 2019-06-19:
Sorry for the late reply
What exactly do you want to do? Re-scale from meters to millimetres?

thesec, 2019-06-20:
Actually I  want that bugged Cura clone to do the scaling, no matter what the units ... just make it scale it to fit the bed ... just like the original - and still so - does.

mteterina, 2019-06-20:
I am sorry but i dont really understand what you mean. Do you have a stl or pictures to explain this issue?

thesec, 2019-06-24:
Throwing in any model in a larger unit than millimeters and larger than the build plate to repro ... sounds dead simple to me.

thesec, 2019-06-24:
Just in case ... having ticked the preference "Scale large models" helps in seeing the bug.

mteterina, 2019-06-26:
Now i understand what you mean, i will forward this issue to our software developer! Thank you!