BCN3D/Cura #145: Cura build infill below roof...

S3D-r, 2019-09-04:
-->

**Application Version**

2.1.5
**Actual Results**

Infill layer will be built below the roof of a horizontal hole

**Expected results**

Roof directly on the top of the hole, no infill layer.

[https://youtu.be/MbUM3GSqMG0](url)

S3D-r, 2019-09-05:
For some reason, Github fails to create a correct hyperlink, another try https://youtu.be/MbUM3GSqMG0

eloipr, 2020-02-27:
This is fixed in future versions of ultimaker cura and in the next releases we are not planning to fix it.

closed 2020-02-27 by eloipr

S3D-r, 2020-02-27:
So, you have reported this to Ultimaker, right? Have they told that it will be fixed or is it just your humble wish... :-) ?

eloipr, 2020-02-27:
Sorry, I think I haven't explained myself well. We haven't reported it. The last Cura version doesn't have this issue, but BCN3D Cura is based on Cura 3.2 (which has the issue). 
Upgrading is a huge amount of work and it's not going to happen soon.

S3D-r, 2020-02-27:
Now that's an accurate and clear answer.

S3D-r, 2020-02-27:
I still wish that each issue would get a flag that it deserves. In this case, I guess the tag would be "a bug".